samyk/poisontap #133: Poisontap need help please

M1ndHunt3r007, 2017-12-08:
Dear,
I am trying to get Poisontap working on my Pi Zero.
I having problems with the backend_server...does any1 knows a good tut for this?
The last 2 days, I was busy searching the webabout this matter, but I cant find any good working tutorial.

Please help

Kind regards.
Natascha

Sorry for my poor English

M1ndHunt3r007, 2017-12-09:
nah...I think this project is dead, waste of time...

samyk, 2017-12-09:
Did you try the example in the readme?

M1ndHunt3r007, 2017-12-09:
Yes, i tried, but no luck.....for the backend serever; I did this:
https://nuxview.blogspot.nl/2016/11/poisontap-setting-up-backend.html
i triedd to search on the net for a more detailed tutorial, but no luck...

samyk, 2017-12-09:
I have no idea what issues you're running into because you've provided no information other than "no luck"

M1ndHunt3r007, 2017-12-09:
[image]

M1ndHunt3r007, 2017-12-09:
i dont know what I must change in the backdoor.html, I have 2 ip's from digital ocean...I dont know which Ip i must put in the backdoor.html

samyk, 2017-12-09:
Are they both public IPs? If you're not sure, what are the IPs? The backend_server should listen to all IPs by default.

M1ndHunt3r007, 2017-12-09:
[image]

M1ndHunt3r007, 2017-12-09:
Which files must i edit now?

samyk, 2017-12-09:
Use 188.166.15.36 - change "YOUR.DOMAIN" in backdoor.html to the IP you want to use

M1ndHunt3r007, 2017-12-09:
Thank you Samyk.

I will work on this matter.....I think i have 1 other issue....:
When I want to do this:
#DHCP: 
sudo cp -f dhcpd.conf /etc/dhcp/dhcpd.conf
add this to  /etc/default/isc-dhcp-server:
INTERFACES="usb0"   (>>>>>I got an error: Permission denied.)
===============

So, I will find out how to edit this...
thanks for helping

M1ndHunt3r007, 2017-12-09:
Thank you Samyk.  
I will work on this matter.....I think i have 1 other issue....: When I want to do this: 
#DHCP:  
sudo cp -f dhcpd.conf /etc/dhcp/dhcpd.conf 
add this to  /etc/default/isc-dhcp-server: 
INTERFACES="usb0"   (>>>>>I got an error: Permission denied)

I will search the web for a solution..
Thnxz for helping

M1ndHunt3r007, 2017-12-09:
This guy did it very good
[https://www.youtube.com/watch?v=BSV-aEkAbNY&t=147s](url)

If some1 can make a tutorial, that is the same as in the video, I will pay him for his time.

Kind regards

M1ndHunt3r007, 2017-12-10:
I got this error message, and I dont know how to fix it:

events.js:163
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE :::1337
    at Object.exports._errnoException (util.js:1050:11)
    at exports._exceptionWithHostPort (util.js:1073:20)
    at Server.setupListenHandle [as _listen2] (net.js:1259:14)
    at listenInCluster (net.js:1307:12)
    at Server.listen (net.js:1406:7)
    at Object.<anonymous> (/poisontap/backend_server.js:50:8)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
root@Poisontap:/poisontap#

samyk, 2017-12-11:
Something else is running on port 1337 (possibly another instance of the backdoor). `lsof -n | grep 1337` should tell you what process and you can `kill -9` it

M1ndHunt3r007, 2017-12-11:
Thnxz, that worked...server is listening on port 1337 now...
thnxz man...

M1ndHunt3r007, 2017-12-12:
Does any1 knows how to host and setup the backend server on Amazon AWS?
Thnxz in advance
I am also willing to pay for the help

kind regards

aidanbowler, 2018-01-22:
hi

aidanbowler, 2018-01-22:
ip  number  for leighton buzzard