samyk/poisontap #135: Backend_server not working

himuura, 2017-12-26:
Hi there @samyk ! First of all, nice work with poisontap, it's really outstanding! Secondly, im having a couple of issues running the backend_server.js on an Amazon EC2 machine. It indeed listens on port 1337, but no connection is established with the infected machine. Trying to use curl, the message gets sent but the infected device does not receive it. I followed this:

$ sudo apt-get install git npm
$ sudo npm install websocket
$ curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash -
$ sudo apt-get install nodejs
$ git clone https://github.com/samyk/poisontap
$ cd poisontap
$ sudo node backend_server.js &
Sat Apr 01 2017 08:30:47 GMT+0000 (UTC) Server is listening on port 1337

Edited the backdoor.html on my pi zero w so it points to my amazon EC2 instance. On the server side, i did not edit any file after installing the necessary packages. My pi zero w is working fine, i can access it locally and inside the poisontap folder there is a cookie.log file. But that doesn't upload to the server, i can only see the cookie file locally. And sending a curl from the server, it doesn't reach the infected machine. I do have a port 1337 opened on the server (amazon security groups, custom TCP rule -> port 1337) but i don't even receive the "hi" from the client. Any idea on what i might be missing?

Thanks in advance!

samyk, 2017-12-26:
I'd check:

- connecting directly to port 1337 to ensure it is open and listening
- watching the Network tab in Inspector/Developer Tools of the browser when hitting the PoisonTap page to see what happens in the connection attempt

himuura, 2017-12-26:
I've tried via browser connecting to my Amazon instance using port 1337 and the server did flare up saying that there was a connection. As for checking the websocket part with the dev tools, haven't tried it yet, gonna check it tomorrow and will give you feedback. Using http://<Amazon instance>:1337 the server throws a message saying that there is a web browser request but nothing more. Is it supposed to? Thanks for the fast reply!

himuura, 2017-12-28:
@samyk tried it today, while the pi zero is connected, i get a lot of ERR_SOCKET_NOT_CONNECTED and after i unplug it, its all 404 not found. no cookie log is exported to the server. Any idea?

ishan-saha, 2018-06-03:
what to do....it is throwing errors

internal/modules/cjs/loader.js:596                                                                                          throw err;                                                                                                              ^                                                                                                                                                                                                                                           Error: Cannot find module 'websocket'                                                                                       at Function.Module._resolveFilename (internal/modules/cjs/loader.js:594:15)                                             at Function.Module._load (internal/modules/cjs/loader.js:520:25)                                                        at Module.require (internal/modules/cjs/loader.js:650:17)                                                               at require (internal/modules/cjs/helpers.js:20:18)                                                                      at Object.<anonymous> (C:\Users\Ishan\Desktop\backend.js:4:23)                                                          at Module._compile (internal/modules/cjs/loader.js:702:30)                                                              at Object.Module._extensions..js (internal/modules/cjs/loader.js:713:10)                                                at Module.load (internal/modules/cjs/loader.js:612:32)                                                                  at tryModuleLoad (internal/modules/cjs/loader.js:551:12)                                                                at Function.Module._load (internal/modules/cjs/loader.js:543:3)